oleg-milantiev/allsky.py #1: Автоматическое удаление устаревших данных

avfc1874, 2019-02-27:
Требуется обеспечить автоматическое удаление устаревших данных.
В системе всегда должны быть доступны данные за последние 7 суток.
Например, ежедневно в 12:00 удаляются данные, которые были получены 8 дней назад.
Время для автоматического удаления желательно иметь возможность настраивать в файле конфигурации.
Реализация данного требования является важной для обеспечения работы в не обслуживаемом режиме МиниПК, который входит в состав Системы "AllSky-камера".

oleg-milantiev, 2019-02-27:
Согласен. Нужно включить в ближайшие планы! Сейчас поищу, как в гитхабе это сделать :)

avfc1874, 2019-03-04:
Привет Олег!
Я немного подумал над реализацией автоудаления и предлагаю реализовать так, чтобы сразу после получения очередного нового файла кадра производить удаление файла кадра недельной давности.
Например:
- сохранен файл кадра 2019-03-04_14-24.jpg	
- удален файл кадра 2019-02-25_14-24.jpg

oleg-milantiev, 2019-03-06:
Решено. В конфиге параметр archive.days указывает сколько суток хранить жпеги в папке snap. Всё выходящее за этот диапазон удаляется.

Проверка после получения каждого кадра.

closed 2019-03-06 by oleg-milantiev