oleg-milantiev/allsky.py #4: Генерить мультик за час

oleg-milantiev, 2019-02-27:
Т.к. ресурсоёмкая задача для мелкой малины, то генерить мультик по запросу пользователя. 
Скорее всего, асинхронно. То есть в фоне.

avfc1874, 2019-02-27:
В конфигурационном файле желательно иметь возможность установить параметр "Количество кадров в секунду" Значение по-умолчанию = 2

oleg-milantiev, 2019-04-15:
Сделал.
Сейчас есть только за час. Но возможности шире.

Нужно проверить соответствие времени. Возможен сдвиг на 3 часа (UTC вместо МСК)

oleg-milantiev, 2019-04-15:
да. Сдвиг был. Устранил

closed 2019-04-15 by oleg-milantiev

avfc1874, 2019-04-15:
Может будет здорово, если Мультик за час будет продолжительностью 1 мин, т.е. 1 сек = 1 мин реального времени?

oleg-milantiev, 2019-04-15:
1 минута = 60 сек. То есть 1fps? Очень медленно для тебя, но норм для тебя - значит нужно дать возможность настраивать частоту кадров.