oleg-milantiev/allsky.py #5: Автоматический подбор экспозиции в зависимости от времени суток

avfc1874, 2019-02-27:
Требуется обеспечить автоматический подбор экспозиции для кадра в зависимости от времени суток, учитываю восход/закат Солнца. Эта информация может быть полезна.
https://askentire.net/q/%D0%9E%D1%82%D0%BA%D1%80%D1%8B%D1%82%D1%8B%D0%B9-api-%D0%B4%D0%BB%D1%8F-%D0%B4%D0%B0%D0%BD%D0%BD%D1%8B%D1%85-%D0%B2%D0%BE%D1%81%D1%85%D0%BE%D0%B4%D0%B0-%D0%B7%D0%B0%D1%85%D0%BE%D0%B4%D0%B0-%D1%81%D0%BE%D0%BB%D0%BD%D1%86%D0%B0-35072728215

oleg-milantiev, 2019-02-27:
Да, но подбор уже есть и функционирует. Конечно, может понадобится разная обработка кадра днём / в сумерках / Лунной / облачной / ясной ночью.
Ну там уровни, гамму покрутить.

oleg-milantiev, 2019-03-27:
Актуально:
http://docs.astropy.org/en/stable/coordinates/solarsystem.html
http://docs.astropy.org/en/stable/api/astropy.coordinates.get_sun.html

closed 2023-10-27 by oleg-milantiev