oleg-milantiev/allsky.py #16: Главная страница - Наличие ссылки на последний mp3-ролик

avfc1874, 2019-03-06:
Обеспечить на главной странице отображение ссылки на последний mp3-ролик, который был создан системой с указанием даты и времени его создания.

avfc1874, 2019-03-07:
исправил mp3 на mp4

oleg-milantiev, 2019-03-07:
К слову сказать, запись звука - очень интересная опция для AllSky. В т.ч. в качестве косвенного детектора дождя / ветра.

Я планировал в своей обсерватории записывать звук под металлической крышей. Использовать его как датчик дождя. Даже мелкие капли очень хорошо детектируются многократным (и длительным) увеличением уровня в микрофоне.

avfc1874, 2019-03-07:
Интересно, но мне кажется избыточно. У меня есть IP-камера со встроенным микрофоном. Лежит в коробочке на даче. Никак не придумаю куда ее поставить :)

oleg-milantiev, 2019-03-17:
Готово

closed 2019-03-17 by oleg-milantiev