oleg-milantiev/allsky.py #18: Вёрстка, красивости

oleg-milantiev, 2019-03-07:
Вёрстка "сайта" allsky камеры. Бутстрап и прочее нужное и современное.

avfc1874, 2019-03-07:
Это хорошее требование, но наверное это Шаблоны страниц:
- Главная
- Авторизация
- Статистика показаний датчиков

oleg-milantiev, 2019-03-07:
Ввиду отсутствия чёткого ТЗ, состав страниц будет пополняться по мере расширения функционала (эволюционный рост). Поэтому, в рамках этой задачи, нужно дать каркас / обёртку / шаблон для всех страниц.

avfc1874, 2019-03-07:
Предложенный концептуальный вариант веб-приложения в виде Панели управления будет удобен.

https://getbootstrap.com/docs/4.3/examples/dashboard/
"...Изначально кадр AllSky на экране. Слева можно выбрать другой раздел, появились архив и данные датчиков. В т.ч. графики..."

oleg-milantiev, 2019-03-07:
Шаблон применил. Остальное по мере развития проекта

closed 2019-03-07 by oleg-milantiev