oleg-milantiev/allsky.py #32: Автоматическое формирование по окончании астроночи суммарного кадра с треками звезд

avfc1874, 2019-03-12:
Желательно по окончании астрономической ночи выполнить автоматическое формирование суммарного кадра с треками звезд и его сохранение. Файл сделать доступным для просмотра в одном из разделов Панели управления.

oleg-milantiev, 2019-03-27:
https://stackoverflow.com/questions/17291455/how-to-get-an-average-picture-from-100-pictures-using-pil
плюс нужно найти границы ночи. Можно по времени, но кмк лучше по фактическому небу со звёздами. После решения https://github.com/oleg-milantiev/allsky.py/issues/38

avfc1874, 2019-03-27:
> 
> 
> https://stackoverflow.com/questions/17291455/how-to-get-an-average-picture-from-100-pictures-using-pil
> плюс нужно найти границы ночи. Можно по времени, но кмк лучше по фактическому небу со звёздами. После решения #38

Логично!