oleg-milantiev/allsky.py #36: "Наклейки" с обозначением сторон горизонта

avfc1874, 2019-03-15:
Желательно иметь возможность в глобальных настройках указать расположение наклеек с-ю-в-з, которые будут сохраняться на кадре.

oleg-milantiev, 2019-03-18:
азимут севера, видимо. Остальные по 90°... или лучше каждую регулировать? Это может понадобиться, если камера установлена не точно вертикально.

avfc1874, 2019-03-18:
В первой версии думаю достаточно будет первого варианта.

oleg-milantiev, 2019-04-09:
Буквы сторон света теперь можно реализовать через config.timestamp, пример тут: https://github.com/oleg-milantiev/allsky.py/blob/master/config.py.dist:16

Нужно добавить 4 аналогичных блока NWES

oleg-milantiev, 2019-04-09:
https://github.com/oleg-milantiev/allsky.py/blob/master/config.py.dist , строка 16

closed 2019-04-09 by oleg-milantiev

oleg-milantiev, 2019-04-09:
Достаточно перезапуска allsky-демона:
https://github.com/oleg-milantiev/allsky.py/wiki/daemon - внизу