oleg-milantiev/allsky.py #44: Архив - переход к времени

oleg-milantiev, 2019-03-24:
Сейчас можно листать кадры. Но если нужно много отлистать, то лучше ниже снимка добавить инструмент перехода к времени снимка. Удобней будет, мне кажется, в виде графика выдержки, например.
А для этого выдержку нужно сохранять: https://github.com/oleg-milantiev/allsky.py/issues/43

avfc1874, 2019-03-27:
Т.е. На графике, который отображает CCD Выдержка будет возможность выбрать временную метку и открыть соответствующий кадр?

oleg-milantiev, 2019-03-27:
Ага, был такой план.
Думаю, как лучше реализовать

avfc1874, 2019-03-27:
Открывается кадр для выбранной позиции. Имеются кнопки Назад - Вперёд с возможностью навигации.

oleg-milantiev, 2019-03-28:
Реализовал в разделе "Графики".
Если выбраны CCD* показатели, то клик на точку на графике открывает новую вкладку с архивом, перемотанным на указанную дату.
В графиках сделал о том подпись (под графиком).

closed 2019-03-28 by oleg-milantiev

avfc1874, 2019-03-28:
Спасибо!
Очень полезная возможность реализована.

Baker1977, 2019-03-28:
Кстати было бы не плохо, если на картинках и на видео писались дата и время

avfc1874, 2019-03-28:
Заслуживает внимания. Лучше к.м.к. оформить это отдельным запросом.

Baker1977, 2019-03-28:
Кстати не очень удобно мне кажется через график на конкретный снимок попадать... эту функцию надо сделать в "Архиве"