oleg-milantiev/allsky.py #48: Железо - ИК датчик

oleg-milantiev, 2019-03-27:
Отдельную герметичную коробку с разъёмом. Быть может обогрев и трубка, сужающая FOV. Надо пробовать.

oleg-milantiev, 2023-03-28:
О, у меня он есть.. С UART конвертором. Не уверен, что до сих пор работает.
Сейчас идея интересней есть! В рамках проекта можно купить 2D ИК сенсор типа MLX90640 или похожий (32х24 пикселей). Конечно, с оптикой вопрос, но если показать AllSky ИК небо... это будет очень... нет ОЧЕНЬ полезно.

[image]

flyeye, 2023-03-28:
У меня такой: 
[image]
На сколько я помню, 15 градусов апертура. 

С другой стороны, у меня оллскай на 485-ой камере, сейчас еще 585-ая вышла, жду доставку. Они очень чувствительные в ИК. Если забить на бб, а может быть даже конвернуть в ч/б - возможно и датчик не нужен?