oleg-milantiev/allsky.py #50: BUG - Видеоролик созданный системой невозможно открыть медиаплеером

avfc1874, 2019-03-28:
Созданный файл 2019-03-27-240p.mp4 невозможно открыть медиаплеером и его просмотреть.

avfc1874, 2019-03-28:
Скачал на локальный диск.
Попытка открыть файл для просмотра медиаплеерами WinAmp, Windows Media Player закончились неудачей. Ранее WinAmp успешно использовался для открытия и просмотра файлов полученных из Системы.

oleg-milantiev, 2019-03-28:
Там mp4, если не ошибаюсь.
Я проигрываю его в VLC. Другими плеерами не пробовал

avfc1874, 2019-03-28:
> 
> 
> Там mp4, если не ошибаюсь.
> Я проигрываю его в VLC. Другими плеерами не пробовал

Странно. Я использую WinAmp с установленной вместе с ним богатой коллекцией видео и аудиокодеков. Этим же плеером вчера смотрел видео предыдущих суток. Пока предлагаю ничего не трогать. Может на ОС рабочего ноута накатили корпоративное обновление? Попробую вечером этот видеоролик открыть дома.

oleg-milantiev, 2019-03-31:
Подтверждаю проблему с ЧБ-роликом на андроиде в хроме.

avfc1874, 2019-03-31:
На компе с ОС Windows аналогичная ситуация. Последние ролики (начиная с 27.03.2019) невозможно просмотреть после их скачивания или плеером, встроенным в браузер.

avfc1874, 2019-04-08:
Спасибо! Исправлено.

closed 2019-04-08 by avfc1874