oleg-milantiev/allsky.py #66: Письмо админу

oleg-milantiev, 2019-04-18:
В перспективе планирую её и видео высылать на почту админу. Кео в виде аттача. Видео - ссылкой.

avfc1874, 2019-04-18:
Не очень понятно, от имени какой учетной записи и какого почтового сервера сообщение электронной почты будет создаваться и отправляться?

oleg-milantiev, 2019-04-18:
Указывается в конфиге. Обычно от яндекс / гугл / мыл.ру учётки. Это не сложно реализовать. Делал много раз в рабочих проектах

oleg-milantiev, 2019-04-22:
Сначала ещё нужно через https://www.astropy.org/ получить время восхода. По нему, я думаю, стоит генерить и мультик, и кео. По нему же отсылать письмо.

oleg-milantiev, 2019-04-22:
https://astroplan.readthedocs.io/en/latest/api/astroplan.Observer.html#astroplan.Observer.sun_rise_time

avfc1874, 2019-04-22:
> 
> 
> https://astroplan.readthedocs.io/en/latest/api/astroplan.Observer.html#astroplan.Observer.sun_rise_time

Ух, какой классный ресурс ресурс нашел с API!
Заслуживает внимания. :)

oleg-milantiev, 2019-04-22:
Не, это не инет-ресурс. Это основная астро-библиотека питона. И её части.
Очень мощная штука, действительно.

avfc1874, 2019-04-22:
> Очень мощная штука, действительно.
Интересно. Какие могут быть потенциальные сценарии использования, которые могут быть рассмотрены на перспективу?

oleg-milantiev, 2019-04-22:
`Интересно. Какие могут быть потенциальные сценарии использования, которые могут быть рассмотрены на перспективу?`

В первую очередь вижу возможность тонко тюнить изображение на основе текущего времени суток. В сумерках нужна такая обработка, а ночью другая. Днём - третья.

oleg-milantiev, 2023-10-27:
Можно добавить рассылку и в telegram через (уже существующего) бота.