oleg-milantiev/allsky.py #79: AllSky Панель администрирования - Перезагрузка миникомпьютера в "ручном" режиме

avfc1874, 2019-05-31:
Требуется обеспечить возможность в разделе "AllSky Панель администрирования" вызвать операцию "Перезагрузка". Результатом этого должна быть перезагрузка AllSky-сервиса или непосредственно миникомпьютера.
Реализация данного требования не потребует от Администратора использовать программу Putty на настольном компьютере или ноутбуке.

flyeye, 2023-03-27:
Поддерживаю. Именно в варианте перезагрузки компьютера. Это нужно для случая, если заглючила камера или ее драйвер. Помогает только аппаратная перезагрузка. А для этого нужно зайти по ssh... Хотелось бы иметь эту возможность из админки.

avfc1874, 2023-03-28:
Предлагаю поднять приоритет реализации этой возможности.

oleg-milantiev, 2023-03-28:
Тут, кажется, не приоритеты, но milestone (этапы). Сейчас сделаю следующий этап, пока не изучал эту возможность. Ну и добавлю туда для теста этот issue.

oleg-milantiev, 2023-03-28:
Добавил план на апрель https://github.com/oleg-milantiev/allsky.py/milestone/9
Можно (в меру разумного) добавлять туда и другие ISSUE

oleg-milantiev, 2023-10-27:
В рамках docker реализации добавил перезагрузку отдельных сервисов. Обычно этого достаточно. Разве что камера зависла.

Буду думать, как теперь перезагрузку через веб безопасно сделать.