oleg-milantiev/allsky.py #81: Сенсоры - Отображение текущих значений T/H/P под Главным меню

avfc1874, 2019-11-03:
Обеспечить под Главным меню отображение текущих значений T/H/P для каждого сенсора, подключенного к системе.

oleg-milantiev, 2019-11-11:
Ввёл в config.py массив sensors. В нём перечислены все сенсоры, подключенные к миникомпу.
Сейчас там только bme280. Перечислены каналы этого сенсора. У тебя, например, это 0 и 1.

Вывел в веб показания всех перечисленных в конфиге сенсоров.

closed 2019-11-11 by oleg-milantiev

avfc1874, 2019-11-11:
> 
> 
> Ввёл в config.py массив sensors. В нём перечислены все сенсоры, подключенные к миникомпу.
> Сейчас там только bme280. Перечислены каналы этого сенсора. У тебя, например, это 0 и 1.
> 
> Вывел в веб показания всех перечисленных в конфиге сенсоров.

Отлично!