oleg-milantiev/allsky.py #84: API - Получение последних значений T/H/P

avfc1874, 2019-11-03:
Требуется обеспечить возможность обработки и предоставления данных по запросу внешнего программного приложения на получение последних значений T/H/P сенсоров, которые подключены к Системе.

avfc1874, 2019-11-04:
В качестве ответа предлагается формировать файл формата XML, следующего содержания:
======
<?xml version="1.0" encoding="utf-8" ?>
<sensors last_update="04.11.2019 15:18:34">
  <sensor sensor_id="1" sensor_name="Под куполом AllSky">
    <param name="temperature" unit="°C" value="14.3" />
    <param name="pressure" unit="мм рт.ст." value="749.4" />
    <param name="humidity" unit="%" value="62" />
  </sensor>
  <sensor sensor_id="2" sensor_name="Снаружи">
    <param name="temperature" unit="°C" value="5.1" />
    <param name="pressure" unit="мм рт.ст." value="749.1" />
    <param name="humidity" unit="%" value="81" />
  </sensor>
</sensors >
======

oleg-milantiev, 2019-11-04:
Не, будет JSON для унификации (т.к. он уже применяется в проекте).

avfc1874, 2019-11-04:
> Не, будет JSON для унификации (т.к. он уже применяется в проекте).
Спасибо! Хорошо.

oleg-milantiev, 2019-11-11:
Готово. Описал в документации: https://github.com/oleg-milantiev/allsky.py/wiki/web-REST

closed 2019-11-11 by oleg-milantiev

avfc1874, 2019-11-11:
> 
> 
> Готово. Описал в документации: https://github.com/oleg-milantiev/allsky.py/wiki/web-REST

Отлично! Спасибо большое!