oleg-milantiev/allsky.py #92: Управление настройками камеры из веб

oleg-milantiev, 2021-11-03:
Перенести настройки камеры из config.py в веб-интерфейс. Чтобы можно было покрутить и посмотреть, что получится. Обязательно с кнопкой возврата к начальным.

flyeye, 2023-03-28:
#96  - входит в эту задачу.

avfc1874, 2023-03-28:
> Обязательно с кнопкой возврата к начальным.

Это весьма полезная возможность!

oleg-milantiev, 2023-03-28:
Есть польза в текстовике config.py. Он, с точки зрения бритвы Оккама, максимально независем от внешних зависимостей. Но, конечно, если вспомнить многочисленные интерфейсы роутеров, ip-камер и другого современного сетевого оборудования, все настройки, так или иначе, должны присутствовать в вебе.

Тогда нужно ВСЕ настройки переносить в веб. Не только какие-то отдельные оставлять в config.py, а какие-то в вебе. Недавно вспомненный Оккам призывает за хранение всех настроек только в одном месте. Видимо, в вебе.

oleg-milantiev, 2023-04-13:
Начал делать

[image]

avfc1874, 2023-04-13:
Понятный интерфейс формы "Настройки"!

avfc1874, 2023-04-24:
> Уф... большой объём изменений. Пока что собрал веб. Ещё не сохраняет. И пользователей пока не редактирует.
> 
> Изучите плиз, вдруг будут какие замечания. Лучше внести корректировки сейчас, если таковые будут.
> 
> https://youtu.be/YJ9sL9--D4w

Здоровски! Понятный интерфейс. Название используемой камеры нужно знать, чтобы драйвера корректно под нее "подхватились" или будет достаточно выбора значения из комбобокса?

oleg-milantiev, 2023-04-24:
> Название используемой камеры нужно знать, чтобы драйвера корректно под нее
> "подхватились" или будет достаточно выбора значения из комбобокса?

На первом этапе через запуск /opt/allsky.py/list.py вручную.
Хочется нарезать фичи покороче. Иначе есть риск захлебнуться и долго браться за доделку.
Она и сейчас достаточно обширна. Однозначно, лучше выносить всё в отдельные issue.

> Может имеет смысл на вкладке Архив установить периодичность очистки данных?

Ага, так и есть сейчас
[image]

oleg-milantiev, 2023-05-04:
Веб готов.
Всё сохраняется в базе в JSON.

Осталось использовать сохранённое в allsky.py

oleg-milantiev, 2023-05-18:
Добавил использование в allsky.py. Случайно в master залил. В следующий раз буду в develop лить. А пока можно начинать тестирвать из master.

Чуть поломал allsky.py в части регулировки обогрева. Хочу вынести это в отдельный скрипт. Пока что под управлением supervisord. В перспективе в docker.

closed 2023-05-18 by oleg-milantiev