oleg-milantiev/allsky.py #97: Plate solving

flyeye, 2023-03-28:
Было бы здорово сделать plate solving и подписывать наиболее яркие звезды и созвездия.

oleg-milantiev, 2023-03-28:
> Какая практическая ценность этого для пользователя AllSky-камеры?

Глянь на камеру САО, например. Интересный вид, но ... порой мне кажется перегруженным инфой. Возможно, должна быть опция показывать ли чистую картинку и / или её обработку.

Это стоит обдумать / обсудить.

flyeye, 2023-03-28:
Однозначно должна быть настройка включать/выключать. 

Польза может быть разная. Созвездия на оллскае неплохо узнаются близко к центру, а чем ближе к краю тем сложнее понять что это за звезда/созвездие. Идентифицировав звезды/созвездия можно а) понимать что закрыто облаками, а что нет, б) на глаз можно грубо определять "проницаемость" в звездных величинах, в) высоту чего-то (серебристых облаков, полярного сияния, метеора) над горизонтом.

avfc1874, 2023-03-28:
> > Какая практическая ценность этого для пользователя AllSky-камеры?
> 
> Глянь на камеру САО, например. Интересный вид, но ... порой мне кажется перегруженным инфой. Возможно, должна быть опция показывать ли чистую картинку и / или её обработку.
> 
> Это стоит обдумать / обсудить.

Не думаю, что AllSky-камеру нужно нагружать несвойственной ей задачей PlateSolv.
 Наверное будет лучше механизм определения Чистого неба/Грязного неба от 10 до 0 ?

avfc1874, 2023-03-28:
> Однозначно должна быть настройка включать/выключать.
>
> Польза может быть разная. Созвездия на оллскае неплохо узнаются близко к центру, а чем ближе к краю тем сложнее понять что это за [...]

В каком виде будет такая информация представлена пользователю? Баллы?

oleg-milantiev, 2023-03-28:
Кстати, я уже реализовал эту задачу локально. Думал о следующем этапе развития Проекта. Код короткий

from astropy.stats import sigma_clipped_stats
from photutils.detection import DAOStarFinder

...
data = hdu.data
...
        mean, median, std = sigma_clipped_stats(data, sigma=2.5)
        daofind = DAOStarFinder(fwhm=2.0, threshold=2.0 * std)

        sources = daofind(data - median)
        print('found stars = {}'.format(sources['xcentroid'].size))

---

В итоге мы имеем:
- кол-во звёзд на снимке (их можно вывести в рамках annotations)
- координаты найденных звёзд (можно сгенерить отдельный jpeg и из него мульт или добавить другой анализатор)

flyeye, 2023-03-29:
> > Однозначно должна быть настройка включать/выключать.
> > Польза может быть разная. Созвездия на оллскае неплохо узнаются близко к центру, а чем ближе к краю тем сложнее понять что это за звезда/созвездие. Идентифицировав звезды/созвездия можно а) понимать что закрыто облаками, а что нет, б) на глаз можно грубо определять "проницаемость" в звездных величинах, в) высоту чего-то (серебристых облаков, полярного сияния, метеора) над горизонтом.
> 
> В каком виде будет такая информация представлена пользователю? Баллы, текстовый Отчет или иное?

Я бы ничего не делал. Никаких отчетов. Просто вспомогательная информация для пользователя. Для оценки качества неба подсчета количества звезд как предлагает Олег вполне достаточно. Ну плюс информация с ИК-датчиков, если добьем эту тему. 

Бывает просто нужно посмотреть чисто ли небо в какой-то части. Или чтобы в какой-то части неба. Не по сторонам света, а в привязке к экваториальной системе координат. Сейчас приходится угадывать измученные дисторсией и прочими аберрациями контуры созвездий. Я бы просто подписывал яркие звезды (альфы, беты, гаммы, до 2-ой звездной величины) и все для начала.