oleg-milantiev/allsky.py #100: Падение allsky.py в newBLOB при работе с БД

flyeye, 2023-03-30:
Иногда (достаточно редко) камера перестает делать снимки. Лог оллская просто обрывается, а вот в логе сепервайзора появляюется вот такие записи: 

Traceback (most recent call last):
  File "/opt/allsky.py/allsky.py", line 200, in newBLOB
    db.commit()
  File "/usr/lib/python3/dist-packages/mysql/connector/connection.py", line 914, in commit
    self._execute_query("COMMIT")
  File "/usr/lib/python3/dist-packages/mysql/connector/connection.py", line 933, in _execute_query
    self.cmd_query(query)
  File "/usr/lib/python3/dist-packages/mysql/connector/connection.py", line 553, in cmd_query
    result = self._handle_result(self._send_cmd(ServerCmd.QUERY, query))
  File "/usr/lib/python3/dist-packages/mysql/connector/connection.py", line 431, in _handle_result
    elif packet[4] == 0:
IndexError: bytearray index out of range

Подозреваю, что дело в потоко- небезопасной работе с курсором. Кроме newBlob он используется в кусочке кода где реализовано управление грелкой. Собственно свои работы по #63 я приостановил как раз потому, что добавил еще один SELECT и падать стало на много чаще.

flyeye, 2023-04-06:
Проблема с потокобезопастностью. Так делать нельзя. Смотрим [тут](https://mysqlclient.readthedocs.io/user_guide.html#mysqldb) и [тут](https://mysqlclient.readthedocs.io/user_guide.html#mysqldb). 

Поискал подтверждение в документации на INDI, но сходу не нашел. Однако уверен, что newBLOB выполняется либо в потоке INDI-сервера/драйвера. Но даже если newBlob выполняется в потоке скрипта, все равно происходит прерывание выполнение тела скрипта. А так как курсор не потокобезопасен (и по второй ссылке написано почему), то при удачном стечении обстоятельств все заканчивается плохо... 

Надо выносить работу с реле в отдельный скрипт, работающий со своим connection`ом.

flyeye, 2023-04-06:
Собственно, второй вариант - использовать критическую секцию при каждом обращении БД. 
[Вот](https://superfastpython.com/thread-mutex-lock/) очень хорошее описание с простыми примерами. Пожалуй поэкспериментирую с этим сначала, вроде совсем просто.

oleg-milantiev, 2023-04-06:
> Надо выносить работу с реле в отдельный скрипт, работающий со своим connection`ом.

В #102 я начал разделение и вынос сервисных задач из allsky.py скрипта получения картинок.
Думаю и stardetect, и анотации тоже там уживутся вместе с watchdog и управлением грелкой, сбором данных с датчиков.

Кстати, в ветке docker перешёл на MySQLdb. Суть примерно та же, но проще установился в ящик на debian:bookworm - https://github.com/oleg-milantiev/allsky.py/blob/c28913f4b38bd46e15ee5a7602edecd97a3c59c7/allsky.py#L17

Возможно, это положительно повлияет на стабильность - он посвежее.

oleg-milantiev, 2023-04-06:
> Собственно, второй вариант - использовать критическую секцию при каждом обращении БД. [Вот](https://superfastpython.com/thread-mutex-lock/) очень хорошее описание с простыми примерами. Пожалуй поэкспериментирую с этим сначала, вроде совсем просто.

Отдельный файл в отдельном потоке с отдельным запуском из-под supervisor / docker, мне кажется само то! Вводить локи тогда и не нужно, так как не будет конкуренции в одном потоке.

oleg-milantiev, 2023-10-27:
Потеряло актуальность при изолировании process.py, работающего с базой от контейнеров INDI и камеры.

closed 2023-10-27 by oleg-milantiev