oleg-milantiev/allsky.py #106: Зацикливание при подборе выдержки

flyeye, 2023-04-07:
Иногда возникает ситуация, когда камера впадает в состояние зацикливания при подборе выдержки. По видимому, проблема в округлении выдержки уже при передачи параметра в камеру.
Например: 
[image]
Камера имеет минимальную выдержку 0.0001. Подозреваю, что 0.00027 она принимает как 0.0003, а 0.00022 как 0.0002. И в итоге прыгает вокруг заданного интервала допустимых значений среднего туда-сюда, хотя интервал вроде бы не маленький... 19000-25000.

oleg-milantiev, 2023-04-07:
Я думаю, нужно ввести ограничение на максимальное кол-во повторов. 10 раз попрыгал туда-сюда, дал кадр. В следующую минуту подберёт лучше.

Основная задача камеры - снимать сцену в условиях небольших плавных изменений освещённости. А не гоняться за самой лучшей экспозицией, отличающейся в разы от прошлой.

flyeye, 2023-04-11:
Сделал, и округление до нужного знака после запятой, и подсчет количества попыток подобрать экспозицию. Пока в ветке CCDGain. Тестирую у себя, после закоммичу и сделаю пулреквест.

oleg-milantiev, 2023-10-27:
В ветке docker полностью (масштабно) переделана процедура подбора выдержки. Зацикливания больше нет.

closed 2023-10-27 by oleg-milantiev