oleg-milantiev/allsky.py #124: Аннотации (наложение текста) - вывод числового значения с плавающей точкой до 5 знаков после запятой

avfc1874, 2023-11-06:
Желательно для Аннотации (наложение текста), которая формируется на основании числового значения с плавающей точкой реализовать вывод на экран до 5 знаков после запятой.

oleg-milantiev, 2023-11-06:
Видимо, ты про вывод exposure. В ней 4 знака норм.
В AI* я уже сократил до 2 знаков.

avfc1874, 2023-11-06:
> Видимо, ты про вывод exposure. В ней 4 знака норм. В AI* я уже сократил до 2 знаков.

Посмотри пожалуйста отображение "CCD avg" по адресу: [http://allsky.vega2017.keenetic.link/](http://allsky.vega2017.keenetic.link/)

oleg-milantiev, 2023-11-09:
А вот гляди. Exp понятно, там нужно 4 знака, т.к. минималка 0.001.
А зачем 4 знака в avg? Мне кажется, там вовсе дробные не имеют смысла.

closed 2023-11-13 by oleg-milantiev